Wiznet/ioLibrary_Driver #3: Integer conversion warning

fritzpas, 2015-04-08:
Hi,

first I will thank you for the great work. ioLibrary_Driver is realy a great lib!

I got a warning in the TI compiler in the socket.c file. Currently it is line 436

else if(tmp & Sn_IR_TIMEOUT)
{
    setSn_IR(sn, Sn_IR_TIMEOUT);
    len = SOCKERR_TIMEOUT;
    break;
}

which ended up in a warning like this

"#69-D integer conversion resulted in a change of sign	socket.c"

I think it is not only a warning, because SOCKERR_TIMEOUT is negative (-13) and len is an unit16_t, so it is a bug.

Kind regards

bingdo, 2015-04-08:
Dear fritzpas

I will make sure this issue.

Thank you for your insights.

closed 2015-04-09 by kaizen8501

midnightcow, 2015-04-09:
Dear fritzpas!!
I have  fixed the issued.
This is closed.
Thank you so much!!!