Wiznet/ioLibrary_Driver #4: Code compatibility problem W5500 <-> W5200 and some more warnings

fritzpas, 2015-04-09:
Hi again,

found another issue in the current master brunch.

My code is running fine with the W5500 chip initialized. If I change the wizchip_conf.h define _WIZCHIP_ to be 5200 my code compiles without any error. But.....

I'm setting the socket interrupt mask register (SIMR) to 0x01 for socket 0 interrupts.
In the current code setSIMR of W5200 is pointing to setIMR2. But this is not correct!

I found out that
W5500 <-> W5200
IR <-> IR
IMR <-> IMR2
SIR <-> IR2
SIMR <-> IMR
Sn_IR <-> Sn_IR
Sn_IMR <-> SnIMR

This means setSIMR() should point to setIMR() and not to setIMR2()!
For SIR <-> IR2 this is correct.
The last problem is that setIMR() should point to setIMR2() but, because of the first assignement this is not possible, isn't it?

The above issue is for set and get functions.

I also get some warnigs that a comma in the typedef enums for e.g. sockint_kind at last entry is not standard! Maybe someone can fix this also.
#230-D trailing comma is nonstandard	socket.h

Kind regards

midnightcow, 2015-04-13:
Hi fritzpas!
Like as you mentioned, I modified the code. 
But, In order to compatible symbol semantics of W5100 & W5500, 
I modified as following
1. setIMR2() & getIMR2()  ==> Set/Get _IMR_ register
2. setIMR() & getIMR()     ==> Set/Get  IMR2 register.

Thank for your interesting in our ioLibrary.

fritzpas, 2015-04-14:
Hi midnightcow,
looks good, but you should remove "& 0xA0" from setIMR2() & getIMR2().

as I told in my previouse post.
setSIMR() of W5500 must be the same as setSIMR of W5200, if not there is no compatibility and I can't simple redefine the _WIZCHIP_.

For example:
The interrupt of socket 1 works with W5500 like this
setSIMR(0x02); // activate interrupt for socket 1
simr = getSIMR(); // chek value after write

For W5200 it were good to use the same code without touching the main()
setSIMR(0x02);
simr = getSIMR();

but what does the current code
setSIMR(0x02) --> setIMR(0x02) -->  WIZCHIP_WRITE(_IMR_, (0x02 & 0xA0))
simr = getSIMR() --> getIMR2() --> (WIZCHIP_READ(_IMR_) & 0xA0)

This is totaly wrong, and the code did'nt work. This must be:
setSIMR(0x02) --> setIMR(0x02) -->  WIZCHIP_WRITE(_IMR_, 0x02)
simr = getSIMR() --> getIMR2() --> (WIZCHIP_READ(_IMR_) )

Then my code works and interrupt function will be called.

Thanks

bingdo, 2015-04-17:
I will check the last issue.
Thanks.

midnightcow, 2015-04-17:
Hi fritzpas!
I did modify the code  as you mentioned.

Thank you.

closed 2015-04-17 by midnightcow