Wiznet/ioLibrary_Driver #7: Multibyte transactions with w5100

fetzerch, 2015-07-04:
Hello!
While integrating your library into my AVR weather station project, I ran into an issue.
I'm not exactly sure if this is a bug, or if it's just a configuration issue.

Currently I struggle even to initialize the W5100 so that it responds to ping.
I was able to track down the problem to a change in [V3.0](1e940bbf14d1e7d4bea84e51416c0230dbf8c9fd) where in [WIZCHIP_WRITE_BUF](https://github.com/Wiznet/ioLibrary_Driver/blob/master/Ethernet/W5100/w5100.c#L132) and [WIZCHIP_READ_BUF](https://github.com/Wiznet/ioLibrary_Driver/blob/master/Ethernet/W5100/w5100.c#L184) the chip select and deselect calls have been moved out of the loop to be beginning and end of the function.

If I move them back into the loop so that the chip gets selected and deselected on every transaction, it suddenly works fine. What I understand from the datasheet is also that this is the correct way for the W5100 and that the multi byte transactions are only supported in the W5200 and later.

Can you confirm this? Or am I missing something?
Hardware wise wise I am using an Aukru W5100 Ethernet Shield and an ATMega328P.

Cheers,
Christian

midnightcow, 2015-07-08:
Hello.
What you mentioned is right, For W5100's multiple transation, Host should be controlled /SCS every byte. ioLibrary_V3.0's w5100 driver has a bug. I will modify the bug and release it ASSP.
Thank for your interesting...

midnightcow, 2015-07-15:
Hello.
I am so sorry for late bug fixed. I was updated V3.0.1.
After test, If you have still problem, Let me know that.
Thank you.

closed 2015-07-15 by midnightcow

fetzerch, 2015-07-24:
Hi! Thanks for the quick fix. It's working well now!
Cheers, Christian