Wiznet/ioLibrary_Driver #14: bug , DHCP fill the  length of HOST_NAME

makc0330, 2017-02-18:
length add  three of more

bingdo, 2017-09-04:
If you look at the source in dhcp.c file, 3 bytes of MAC address is inserted. You can search that by hostname string.
	pDHCPMSG->OPT[k++] = DHCP_CHADDR[3];
	pDHCPMSG->OPT[k++] = DHCP_CHADDR[4];
	pDHCPMSG->OPT[k++] = DHCP_CHADDR[5];
	pDHCPMSG->OPT[k - (i+3+1)] = i+3; // length of hostname
I guess that is a delimiter. Of course, I think char type would be good in this case.

closed 2017-09-04 by bingdo

caiwei33, 2017-09-12:
不是说你的type有问题，你有没有想过你为啥要加3尼？why could you add three of length of hostname， 前面使用for 填充hostname ，为什么后面长度你要多加3？ 你加了之后根本就不是hostname的长度了。

bingdo, 2017-09-14:
Sorry. I don't know if I understood your Chinese correctly.
Hostname and length is decided by each target included WIZnet chip or module. All target can be distinguished by added 3 bytes(MAC info).
If you want to inquire about other things, please write in English.
Thanks.

caiwei33, 2017-09-14:
you mean that  add dhcp_chaddr bytes to hostname deliberately?
but in ascii code it is maybe the mess code or make router  to parse the hostname is fail ,
because  the hostname macro is character set with end of zero ,you append  other bytes to marco of end