Wiznet/ioLibrary_Driver #17: Network_init(); not setting the Mac, IP, Subnet , Gw, DNS

ToasTer86, 2017-10-18:
while using the network_init(); function the mac, ip, subnet, gw, dns do not get set while using this library.

`void network_init(void)
{
  uint8_t tmpstr[6];
	ctlnetwork(CN_SET_NETINFO, (void*)&gWIZNETINFO);	
	ctlnetwork(CN_GET_NETINFO, (void*)&gWIZNETINFO);
	
	// Display Network Information
	ctlwizchip(CW_GET_ID,(void*)tmpstr);
	printf("\r\n=== %s NET CONF ===\r\n",(char*)tmpstr);
	printf("MAC: %02X:%02X:%02X:%02X:%02X:%02X\r\n",gWIZNETINFO.mac[0],gWIZNETINFO.mac[1],gWIZNETINFO.mac[2],
		  gWIZNETINFO.mac[3],gWIZNETINFO.mac[4],gWIZNETINFO.mac[5]);
	printf("SIP: %d.%d.%d.%d\r\n", gWIZNETINFO.ip[0],gWIZNETINFO.ip[1],gWIZNETINFO.ip[2],gWIZNETINFO.ip[3]);
	printf("GAR: %d.%d.%d.%d\r\n", gWIZNETINFO.gw[0],gWIZNETINFO.gw[1],gWIZNETINFO.gw[2],gWIZNETINFO.gw[3]);
	printf("SUB: %d.%d.%d.%d\r\n", gWIZNETINFO.sn[0],gWIZNETINFO.sn[1],gWIZNETINFO.sn[2],gWIZNETINFO.sn[3]);
	printf("DNS: %d.%d.%d.%d\r\n", gWIZNETINFO.dns[0],gWIZNETINFO.dns[1],gWIZNETINFO.dns[2],gWIZNETINFO.dns[3]);
	printf("======================\r\n");
}
`

tmtpune, 2017-10-18:
I am facing same problem.......Sometimes Network_init Succeds and most of the time FAILS to assign mac, ip, subnet, gw.........Surprisingly DNS assignment succeeds.....

Any help is appreciated

tmtpune, 2017-10-18:
wiz_NetInfo gWIZNETINFO = { .mac = {0x00, 0x08, 0xdc,0x52, 0xb6, 0xec},
                            .ip = {192, 168, 2, 50},
                            .sn = {255,255,255,0},
                            .gw = {192, 168, 2, 1},
                            .dns = {203,187,215,35},//dns = {203,187,215,66
                            .dhcp = NETINFO_STATIC };//NETINFO_DHCP

void network_init(void)
{
   uint8_t tmpstr[6];
   
   SEGGER_RTT_printf(0,"\r\n=== %s NET CONF ===\r\n",(char*)tmpstr);
	SEGGER_RTT_printf(0,"MAC: %02X:%02X:%02X:%02X:%02X:%02X\r\n",gWIZNETINFO.mac[0],gWIZNETINFO.mac[1],gWIZNETINFO.mac[2],
		  gWIZNETINFO.mac[3],gWIZNETINFO.mac[4],gWIZNETINFO.mac[5]);
	SEGGER_RTT_printf(0,"SIP: %d.%d.%d.%d\r\n", gWIZNETINFO.ip[0],gWIZNETINFO.ip[1],gWIZNETINFO.ip[2],gWIZNETINFO.ip[3]);
	SEGGER_RTT_printf(0,"GAR: %d.%d.%d.%d\r\n", gWIZNETINFO.gw[0],gWIZNETINFO.gw[1],gWIZNETINFO.gw[2],gWIZNETINFO.gw[3]);
	SEGGER_RTT_printf(0,"SUB: %d.%d.%d.%d\r\n", gWIZNETINFO.sn[0],gWIZNETINFO.sn[1],gWIZNETINFO.sn[2],gWIZNETINFO.sn[3]);
	SEGGER_RTT_printf(0,"DNS: %d.%d.%d.%d\r\n", gWIZNETINFO.dns[0],gWIZNETINFO.dns[1],gWIZNETINFO.dns[2],gWIZNETINFO.dns[3]);
	SEGGER_RTT_printf(0,"======================\r\n");
	
	ctlnetwork(CN_SET_NETINFO, (void*)&gWIZNETINFO);
	
	
	
	ctlnetwork(CN_GET_NETINFO, (void*)&gWIZNETINFO);

	// Display Network Information
	ctlwizchip(CW_GET_ID,(void*)tmpstr);
	//SEGGER_RTT_printf(0,"\r\n=== %s NET CONF ===\r\n",(char*)tmpstr);
	//SEGGER_RTT_printf(0,"MAC: %02X:%02X:%02X:%02X:%02X:%02X\r\n",gWIZNETINFO.mac[0],gWIZNETINFO.mac[1],gWIZNETINFO.mac[2],
		  //gWIZNETINFO.mac[3],gWIZNETINFO.mac[4],gWIZNETINFO.mac[5]);
	//SEGGER_RTT_printf(0,"SIP: %d.%d.%d.%d\r\n", gWIZNETINFO.ip[0],gWIZNETINFO.ip[1],gWIZNETINFO.ip[2],gWIZNETINFO.ip[3]);
	//SEGGER_RTT_printf(0,"GAR: %d.%d.%d.%d\r\n", gWIZNETINFO.gw[0],gWIZNETINFO.gw[1],gWIZNETINFO.gw[2],gWIZNETINFO.gw[3]);
	//SEGGER_RTT_printf(0,"SUB: %d.%d.%d.%d\r\n", gWIZNETINFO.sn[0],gWIZNETINFO.sn[1],gWIZNETINFO.sn[2],gWIZNETINFO.sn[3]);
	//SEGGER_RTT_printf(0,"DNS: %d.%d.%d.%d\r\n", gWIZNETINFO.dns[0],gWIZNETINFO.dns[1],gWIZNETINFO.dns[2],gWIZNETINFO.dns[3]);
	//SEGGER_RTT_printf(0,"======================\r\n");
}

Not many changes are done to the code.  One strange thing is that except DNS everything(mac,ip,subm,gw) is set to 0.0.0.0

bingdo, 2017-10-19:
Dear ToasTer86
In my experience there is a problem with the interface between host and W5100 chip.
Which interface do you use? SPI or Bus?
Please check the interface and then upload a topic in forum.wiznet.io site.
If you need to confirm your H/W, please send schematic, gerber files to support@wiznet.io

Dear tmtpune
First of all, I think you have to check the interface between host and chip too.
What kind of chip and interface do you use?
You can upload your topic in forum.wiznet.io site too.

Thanks

closed 2017-10-19 by bingdo

tmtpune, 2017-10-19:
Hi bingdo,

I was trying to replicate the set up given in the following link:

http://wiznetmuseum.com/portfolio-items/ble-to-ethernet-thin-gateway/

connections between NRF51 and wiznet board are done as mentioned in the link.

I am using the same ioLibrary_Driver. Is not consistently able to assign GW, SUBM, IP, & MAC

But with DNS it has no problem. I am not able to figure out if the problem is with SET or GET

ctlnetwork(CN_SET_NETINFO, (void*)&gWIZNETINFO);

OR

ctlnetwork(CN_GET_NETINFO, (void*)&gWIZNETINFO);

GET gives me all zeros(00:00:00:00:00:00, 0.0.0.0) except for DNS. DNS assigns successfully

ToasTer86, 2017-10-19:
Hey Bingdo,

I am using SPI (spi0). I have the same issue as tmtpune, the dns is being set but the mac, ip and gw are not being set. They all return 0.0.0.0.

So you say I have to make a post on the wiznet forum. I will do this. 

Toaster

ToasTer86, 2017-10-19:
tmtpune, should we exchange contact details so we can stay up to date with each other how we resolve it etc.

tmtpune, 2017-10-19:
Hi ToasTer86,
sudhish.vandavasi@threemediatech.com
This is my mail id...
DO let me know if u find any solution...

ToasTer86, 2017-10-19:
Hi tmtpune, 

My email is mark.ehrhart@altran.nl You can also add me on linq/skype on that email.

same to you, I am verifying the output now through a logic analyzer.

tmtpune, 2017-10-19:
Hi tmtpune,
I do not know why..... but every mail i send  u on ....."mark.ehrhart@altran.nl"  BOUNCES BACK