Wiznet/ioLibrary_Driver #18: W5500 : Is not consistently able to assign GW, SUBM, IP, & MAC

tmtpune, 2017-10-19:
I was trying to replicate the set up given in the following link:

http://wiznetmuseum.com/portfolio-items/ble-to-ethernet-thin-gateway/

connections between NRF51 and wiznet board are done as mentioned in the link.

I am using the same ioLibrary_Driver. Is not consistently able to assign GW, SUBM, IP, & MAC

But with DNS it has no problem.  I am not able to figure out if the problem is with SET or GET

ctlnetwork(CN_SET_NETINFO, (void*)&gWIZNETINFO);

OR 

ctlnetwork(CN_GET_NETINFO, (void*)&gWIZNETINFO);

GET gives me all zeros(00:00:00:00:00:00, 0.0.0.0) except for DNS. DNS assigns successfully

tmtpune, 2017-10-19:
BY inconsistency i mean :    sometimes w5500 is initialized and sometimes not. i.e static ip, mac, gw, subm get assigned and sometimes fails. failure rate is high.(success:failure) = (1:50)

BUT DNS initialization always succeeds

ToasTer86, 2017-10-19:
For my issue it was changing the SPI mode, I was using the Wiz5100 and had to change the SPI mode to 0

tmtpune, 2017-10-19:
HI ToasTer86,
DId u check this link....
https://forum.wiznet.io/t/topic/2323

Do let me know if u figure out something.......

tmtpune, 2017-10-19:
Hi ToasTer86,

"For my issue it was changing the SPI mode, I was using the Wiz5100 and had to change the SPI mode to 0"  what did u mean by this..........did changing SPI mode to 0  solve your issue

ToasTer86, 2017-10-19:
Hey tmtpune,

I was able to fix the issue. The Wiznet 5100 does not support SPI mode 3, only SPI mode 0. The library we are both using has the SPI0 interface initiate on SPI mode 3.

So after I changed the SPI mode to 0 everything started working. 
[image]

My IP adress is being setup correctly and I can ping the Wiznet 5100 as well.

Which chip are you using? If it is the Wiznet 5100 be sure to check you got the SPI mode on 0

void spi0_master_init()
{
    SPIConfig_t spi_info = {.Config.Fields.BitOrder = SPI_BITORDER_MSB_LSB,
                        .Config.Fields.Mode     = **SPI_MODE0**,
                        .Frequency              = SPI_FREQ_8MBPS,
                        .Pin_SCK                = ARDUINO_13_PIN,
                        .Pin_MOSI               = ARDUINO_11_PIN,
                        .Pin_MISO               = ARDUINO_12_PIN,
                        .Pin_CSN                = ARDUINO_10_PIN};	
    spi_master_init(SPI0,&spi_info);
}

tmtpune, 2017-10-19:
HI ToasTer86,
I am using Wiznet 5500.....Let me check if this solution of yours, solves my problem........
I will be able to revert back with the result only on monday.........
Thank u

ToasTer86, 2017-10-19:
good luck

tmtpune, 2017-10-19:
u r possibly right......as far as i remember at one point of time while merging the code i came across this file: nrf_drv_config.h in which the following can be found
#define SPI0_ENABLED 1
#define SPI1_ENABLED 0
#define SPI2_ENABLED 0

ToasTer86, 2017-10-19:
That is the SPI bus you are using on your primary board ( SPI master) . SPI also had 4 different modes in which it communicates : 0, 1, 2 and 3 
Read more about here: https://www.corelis.com/education/SPI_Tutorial.htm
[image]

The wiznet 5100 only operates on SPI mode 0, and I have it configure to SPI bus 0 on my main board (nRF52840).

tmtpune, 2017-10-19:
Hi ToasTer86,
Can you send me the connection diagram of nrf52840 and wiznet 5100.

I have to create a similar setup using nrF52840 and wiznet5500 (WIZ550io) ........
Current setup comprises nrf51 and wiznet5500 (WIZ550io)
i have used the help of http://wiznetmuseum.com/portfolio-items/ble-to-ethernet-thin-gateway/ 
connect these boards

bingdo, 2017-10-19:
Dear all

Do you know the following site?
Maybe it will help.
[](http://wizwiki.net/wiki/doku.php?id=oshw_using_wiznet:bletoethernet)
[](https://devzone.nordicsemi.com/blogs/1065/wiznets-ble-to-tcp-gateway/)

First, please check H/W connection by reference schematic.
Second, make sure SPI interface between host and W5100/W5500 chip in firmware level.

Thanks.

closed 2017-10-19 by bingdo

tmtpune, 2017-10-20:
Dear all,
i would like to share my new observation.......
The network initialization always succeeds in the case of a recently programmed host(nrf51). But when i restart the host network initialization fails. But if i reprogram(flash) the host with the same .hex file, network initialization succeds.......what could be the reason