Wiznet/ioLibrary_Driver #19: w550io + nrf51(pca10028) : Network initialization succeeds until first restart of host after it has been freshly flashed(programmed)

tmtpune, 2017-10-20:
Dear all,
i would like to share my new observation.......
The network initialization always succeeds in the case of a recently programmed host(nrf51). But when i restart the host, network initialization fails.

But if i reprogram(flash) the host with the same .hex file, network initialization succeds.......what could be the reason

tmtpune, 2017-10-20:
The following image should give you an idea of the scenario.

[image]

tmtpune, 2017-10-20:
The following image should give an idea about the HW connections
[image]

bingdo, 2017-10-24:
I recommend you post the content on the forum. [forum.wiznet.io](url)
It will be difficult to hear the answer to your application here.

Thanks.

closed 2017-10-24 by bingdo