Wiznet/ioLibrary_Driver #22: Use of Malloc

lonpalmer, 2017-11-27:
httpServer.c uses Malloc to store strings.  In smaller embedded targets (like AVR), malloc is problematic.  Could we get rid of the malloc in favor of a fixed length string buffer or perhaps allow a #define? 

In httpServer.c line 678.

bingdo, 2017-11-28:
You can refer to source code of the following link.
[](https://github.com/Wiznet/WIZ550web/tree/master/WIZ550web_Firmware/WIZ550web_HW_Rev1.1/WIZ550web_App/src/Internet/httpServer)

Thanks.

bingdo, 2017-11-28:
You can refer to source code of the following link.
[WIZ550web](https://github.com/Wiznet/WIZ550web/tree/master/WIZ550web_Firmware/WIZ550web_HW_Rev1.1/WIZ550web_App/src/Internet/httpServer)

Thanks.

closed 2017-11-28 by bingdo

lonpalmer, 2017-12-09:
Sorry, i don't understand your point. The source is still using malloc.

bingdo, 2017-12-11:
Did you check source codes of http server in WIZ550web_App project?
I think there is no use case of malloc.
If you have more question about http server, please upload your topic in [forum.wiznet.io](https://forum.wiznet.io/) site.

Thanks.

lonpalmer, 2017-12-11:
Yes, I just confirmed it's still there in this project:

https://github.com/Wiznet/ioLibrary_Driver/blob/master/Internet/httpServer/httpServer.c

Same Line I stated earlier.

Why is the httpServer in this project and in the WIZ550web_App project?

bingdo, 2017-12-11:
Refer to the following.
[https://github.com/Wiznet/WIZ550web/tree/master/WIZ550web_Firmware/WIZ550web_HW_Rev1.1/WIZ550web_App/src/Internet/httpServer](https://github.com/Wiznet/WIZ550web/tree/master/WIZ550web_Firmware/WIZ550web_HW_Rev1.1/WIZ550web_App/src/Internet/httpServer)

WIZ550web is a http server project.

lonpalmer, 2017-12-12:
I see. Why does this project also have http server code?

https://github.com/Wiznet/WIZ550web/tree/master/WIZ550web_Firmware/WIZ550web_HW_Rev1.1/WIZ550web_App/src/Internet/httpServer<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FWiznet%2FWIZ550web%2Ftree%2Fmaster%2FWIZ550web_Firmware%2FWIZ550web_HW_Rev1.1%2FWIZ550web_App%2Fsrc%2FInternet%2FhttpServer&data=02%7C01%7C%7Cc0c65fd3ba2444315e1608d540edba6f%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636486312128078534&sdata=mduFBM%2FiZc8FyTtsZqp1L1H1Utu0WX%2Fu%2FP1v%2BMrSovs%3D&reserved=0>

bingdo, 2017-12-13:
If you need more info of WIZ550web, refer to the following.
[http://wizwiki.net/wiki/doku.php?id=products:wiz550web:start](http://wizwiki.net/wiki/doku.php?id=products:wiz550web:start)
Thanks.