Wiznet/ioLibrary_Driver #26: How to compile w5500 driver 通

YumingMa, 2017-12-25:


renakim, 2018-07-05:
Unfortunately, the android kernel does not include the W5500 driver.

If you want to use the W5500, you have to add the driver code to the kernel.
Please refer to the below link.
* [https://github.com/borg42/w5x00](https://github.com/borg42/w5x00)

Thank you.

closed 2018-07-05 by renakim

renakim, 2018-07-05:
Hi,
The w5500 driver has been added since the official Linux kernel 4.7 or later.

If the Android kernel version is 4.7 or higher, it may be supported. 
Please check this and refer to the below link.

* [Linux kernel commit history: support W5500](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=35ef7d689d7d54ab345b179e50c749fe3a2529eb)

Thank you.