Wiznet/ioLibrary_Driver #31: flush udp buffer

gpzhaome, 2018-02-03:
Hi all, is the flush/clear/refresh udp socket buffer function implemented in the library? I couldn't find it. I'm using W5500 to communicate with a realtime controller. It is important to flush the buffer regularly. 

Thanks.

midnightcow, 2018-02-20:
Hi Dears! 
W5500/W5100/W5200 can flush the data of socket buffer. 
How to flush is just Sn_RX_RD updated without data copy and commanded with Sn_CR[RECV].

int_t sock_flush(uint8_t s)
{
       int_t ret = 0;
       if(getSn_SR(s) == SOCK_CLOSED) return -1;
       ret = getSn_RX_RSR(s);
       setSn_RX_RD( s, (getSn_RX_RD(s)+ret);
       return ret;
}

P.S>
W5300 can not flush the data of socket.  
To flush the data, 
In case of UDP & IPRAW socket, Just reopen a socket.
In case of TCP, There is no flush method.

Thank you.

closed 2018-02-20 by midnightcow

gpzhaome, 2018-02-23:
Great. Thanks a lot.