Wiznet/ioLibrary_Driver #46: Fix in DNS client

bart112233, 2018-05-24:
Need add reset to zero vars "retry_count" and "dns_1s_tick" everytime on DNS_init call.

```
diff --git a/Internet/DNS/dns.c b/Internet/DNS/dns.c
index d4e5806..b412fc2 100644
--- a/Internet/DNS/dns.c
+++ b/Internet/DNS/dns.c
@@ -120,6 +120,7 @@ uint8_t  DNS_SOCKET;    // SOCKET number for DNS
 uint16_t DNS_MSGID;     // DNS message ID
 
 uint32_t dns_1s_tick;   // for timout of DNS processing
+static uint8_t retry_count;
 
 /* converts uint16_t from network buffer to a host byte order integer. */
 uint16_t get16(uint8_t * s)
@@ -472,7 +473,6 @@ int16_t dns_makequery(uint16_t op, char * name, uint8_t * buf, uint16_t len)
 
 int8_t check_DNS_timeout(void)
 {
-	static uint8_t retry_count;
 
 	if(dns_1s_tick >= DNS_WAIT_TIME)
 	{
@@ -496,6 +496,8 @@ void DNS_init(uint8_t s, uint8_t * buf)
 	DNS_SOCKET = s; // SOCK_DNS
 	pDNSMSG = buf; // User's shared buffer
 	DNS_MSGID = DNS_MSG_ID;
+	retry_count = 0;
+	dns_1s_tick = 0;
 }
 
 /* DNS CLIENT RUN */
```

bart112233, 2018-05-24:
or perhaps it is better to do this in the beginning DNS_run

khj098765, 2018-07-05:
We will update after review.
thank you very much

closed 2018-07-05 by khj098765