Wiznet/ioLibrary_Driver #48: DHCP client stops working sometimes

jiripaar, 2018-07-02:
Hello,
I am using your DHCP client source files. The problem is that DHCP client stops working sometimes. The program still waits if the socket is closed (file: socket.c, line: 193 - `while(getSn_SR(sn) == SOCK_CLOSED);` ).
I would understand that this problem can happen when the ethernet cable is disconnected (but this situation should be caught). But this situation happends when the cable is connected for all the time.

Thank you for your help.

khj098765, 2018-07-05:
this problem is not related ethernet cable.
I have some questions.
What type of our product you are using and what interface to use with your mcu?
and... what kind of application project you use now?

jiripaar, 2018-07-09:
Hello,
Sorry, I have forgot to write these important information.
I am using W5500 ethernet controller, I would like to use this chip with NRF52832 microprocessor. This microprocessor has implemented Bluetooth communication. My goal is to able to set parameters over the Bluetooth and over the ethernet.
But I think that the problem is in hardware connection. At this moment I am using the W5500-EVB kit with development kit with NRF52832 and the connection via cables isn't stable. The W5500 is reset after several minutes itself (all data is clear - IP, mask, GW, MAC, ... everything is zero).

khj098765, 2018-07-09:
Hi 
You can refer to the BLE Board created by WIZnet.
https://wizwiki.net/wiki/doku.php?id=oshw_using_wiznet:bletoethernet

reference schematic  : nrf52832 & W5500
https://drive.google.com/open?id=0Bx-BD_H8XJXxVWdrcVliY1E1RGM

example firmware : nrf52832 & W5500
https://github.com/Wiznet/WIZBLE510_ETHERNET

closed 2018-09-09 by khj098765