Wiznet/ioLibrary_Driver #54: -Wsizeof-pointer-memaccess memset(pdhdr, 0, sizeof(pdhdr));

newAM, 2018-09-02:
This line generates an error when compiling, needs to be dereferenced.

https://github.com/Wiznet/ioLibrary_Driver/blob/15b79ef7c0375d7c698e41cfd2dbe9704a9d68c7/Internet/DNS/dns.c#L343

khj098765, 2018-09-09:
I didn't occurred error.
What is the error in the environment?

newAM, 2018-09-10:
```
Internet/DNS/dns.c:343:25: warning: argument to 'sizeof' in 'memset' call is the same expression as the destination; did you mean to dereference it? [-Wsizeof-pointer-memaccess]
  memset(pdhdr, 0, sizeof(pdhdr));
```

This is the exact printout, using gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.10)

I dereferenced pdhr in the sizeof argument like this
`memset(pdhdr, 0, sizeof(*pdhdr)); `
and no longer get the warning

closed 2018-10-17 by khj098765