Wiznet/ioLibrary_Driver #56: See I have a problem with the code display or download, for ioLibrary_Driver/Ethernet/wizchip_conf.h

arikadogenya, 2018-09-18:
line 36-line 44
//! LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR 
//! CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF 
//! SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
//! INTERRUPTION) HOWEVER CAUSED AND#ifdef __cplusplus
extern ¡°C¡± {
#endif ON ANY THEORY OF LIABILITY, WHETHER IN 
//! CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
//! ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF 
//! THE POSSIBILITY OF SUCH DAMAGE.

//! is missing for some lines

arikadogenya, 2018-09-18:
please check it.

irinakim12, 2018-09-27:
@arikadogenya 
I fixed it.  
Thank you

closed 2018-09-27 by irinakim12