Wiznet/ioLibrary_Driver #69: fix ifdef and brackets in Internet/DNS/dns.c

fackgithab, 2018-11-21:
small issue with _DNS_DEUBG_ and printf

fackgithab, 2018-11-21:
can't attach diff on github
diff --git a/Internet/DNS/dns.c b/Internet/DNS/dns.c
index 9f7107b..449ac0a 100644
--- a/Internet/DNS/dns.c
+++ b/Internet/DNS/dns.c
@@ -367,8 +367,8 @@ int8_t parseDNSMSG(struct dhdr * pdhdr, uint8_t * pbuf, uint8_t * ip_from_dns)
 	for (i = 0; i < pdhdr->qdcount; i++)
 	{
 		cp = dns_question(msg, cp);
-   #ifdef _DNS_DEUBG_
-      printf("MAX_DOMAIN_NAME is too small, it should be redfine in dns.h"
+   #ifdef _DNS_DEBUG_
+      printf("MAX_DOMAIN_NAME is too small, it should be redfine in dns.h");
    #endif
 		if(!cp) return -1;
 	}
@@ -377,8 +377,8 @@ int8_t parseDNSMSG(struct dhdr * pdhdr, uint8_t * pbuf, uint8_t * ip_from_dns)
 	for (i = 0; i < pdhdr->ancount; i++)
 	{
 		cp = dns_answer(msg, cp, ip_from_dns);
-   #ifdef _DNS_DEUBG_
-      printf("MAX_DOMAIN_NAME is too small, it should be redfine in dns.h"
+   #ifdef _DNS_DEBUG_
+      printf("MAX_DOMAIN_NAME is too small, it should be redfine in dns.h");
    #endif
 		if(!cp) return -1;
 	}

midnightcow, 2019-01-29:
Refer to Fixed Issue 69

closed 2019-01-29 by midnightcow