Wiznet/ioLibrary_Driver #81: Passive FTP not working

buehlerf, 2019-05-02:
I am testing the FTP server of the W5500.  

Active FTP-Mode works fine. 

However, passive FTP has a problem.  The setup of the connection works fine, but when the code goes to "case STOR_CMD" the connect call tries to open a connection to remote_ip 0.0.0.0 (Line 872 of ftpd.c): Hence the passive FTP implementation does not store the IP address of the FTP client before connecting the data socket.

To reproduce you can use the following python lines:
`import ftplib
session = ftplib.FTP('192.168.10.10','abc','123')
session.set_pasv(False) #works
#session.set_pasv(True) #does not work
file = open('homer.png','rb')
session.storbinary('STOR homer.png', file, 512)
file.close()
session.quit()`

WIZbecky, 2019-11-20:
Hello,

I modified FTP Server code.
So, please check it working well.
And you need to input user and password.

WIZbecky, 2019-11-21:
In passive mode, FTP Client try to connect and open data socket. It is not FTP Server. Please check it.

closed 2019-12-03 by midnightcow