Wiznet/ioLibrary_Driver #94: socket.c infinite loop

Tomsiik, 2020-01-08:
In functions _socket()_ and _close()_ is an infinite loop for checking Sn_CR register when W5100 not responds or it not connected to MCU.

_while( getSn_CR(sn) );_

I solved it by adding a check of repeat attempts.

_uint8_t repeat = 0;
	do {
		if (repeat > 10) {
			return SOCKERR_TIMEOUT;
		}
		repeat++;
	} while (getSn_CR(sn));_

zhigangbox, 2020-09-26:
这里确实是个问题，RT-THread的wiz软件包基于V1.0.3版本，新建的socke（前端无路由器，直接接在交换机上）居然不是closed状态而是0x22，造成直接-1返回造成一直无法通讯。大家有这个问题可以讨论QQ2730122869

hp-beko, 2020-09-26:
Hi everybody,

@Tomsiik is right. infinite loop is a no go in bare metal programming.
@zhigangbox Could you please write in englisch?

Best regards
Hannes

ApmTIMEmqA, 2021-02-02:
I had the same symptoms in W5300 with 28335, and I resolved it as follows.

```C
#define getSn_CR(sn) \
   ((uint8_t)WIZCHIP_READ(Sn_CR(sn)) & 0xFF)
```

taylorh140, 2024-05-06:
I ran into this problem using the standard STM32 library. I think the issue for me was that i had the Recive buffer for the single byte on the stack. EG. 
i used the "HAL_SPI_Receive"
I changed this to use a static value for the receive byte and now it works fine.