Wiznet/ioLibrary_Driver #114: Packet size in MQTT

kaminj, 2021-07-28:
Hello
I’m using IoLibrary and I have this problem.
I’m using w5500 as my chip and I’m a mqtt client who is using iolibrary.
Until packet size is less than my socket dedicated volume everything is ok . I can subscribe and get message successfully .
But when packet size is more than socket dedicated volume, I receive nothing despite of subscribing this special topic.
Maximum socket dedicated volume is 16kb in w5500 chip, but my desired data size is more than 16kb its about 40kb .

Please help me solve this problem . what should I do ?
Thank you.

Nefrin, 2021-07-30:
According to the [datasheet](https://wizwiki.net/wiki/lib/exe/fetch.php/products:w5500:w5500_ds_v109e.pdf) the w5500 has only 32Kb of memory. 
I don't know the exact behavior on a buffer overflow but I guess packets above 32Kb just get dropped.

kaminj, 2021-08-01:
Thanks Hans.